nygma2004/growatt2mqtt #1: only one datapoint is in ioBroker

DraycD, 2022-01-17:
Hi,
first of all thank you for all your work.

i have the issue that in iobroker is only one datapoint were is all in, but not seperate points:
do you have any idea how it can send seperate?
[image]

nygma2004, 2022-01-25:
I designed it to send all the data into a single topic in JSON format, because it is much easier to process for me rather than having 30 messages all in separate topic. If you want that, you need to re-write the `ReadInputRegisters()` method. There is a section with a comment saying `Generate the modbus MQTT message`. As you can see I put all the data into a JSON there. You can break that up, and send each of them into a separate MQTT topic. 
I hope that will be enough, as there will be a lot of MQTT communication done in that method. It may take too long time and the ESP may crash not having enough time to perform other housekeeping tasks.

SHODK, 2022-02-05:
Hi, 
What a great project you have made here, I follow you video and think I made a correct setup, however I get Error: Invalid slave ID on the MQTT
My Growatt is smae as yours, MIN 3000TL-XE 
I check the COM ID on Growatt it is set to 001
I try change it to 002 or 003 with the same change in Arduino sketch
I double check my wiring of Wemos D1 and TTl to RS-485 module, I have same 230V power supply as you show
No matter what I try it give me same error, do you have any idea where to look?

<img width="368" alt="Arduino-Growatt" src="https://user-
[image]
images.githubusercontent.com/92098787/152648230-80e9e74a-ebfb-4a87-84d6-b01c47f0c341.png">

DraycD, 2022-03-04:
I split it now direct in iOBroker.
Is there a Json path in browser?
something like http://192.168.2.43/cm?cmnd=status%208

sorry i m a noob

AlexanderSch90, 2023-03-21:
Hey @DraycD , can you please share your Iobroker code (your javascript; your blockly) to split the json?

hendrikhoffmann, 2024-02-11:
I forked the project and changed the mqtt transfer from json to single data points.
I also added a Webinterface and API for controlling the inverter limits.
see https://github.com/hendrikhoffmann/growatt2mqtt

nygma2004, 2024-02-13:
> I forked the project and changed the mqtt transfer from json to single data points. I also added a Webinterface and API for controlling the inverter limits. see https://github.com/hendrikhoffmann/growatt2mqtt

Thanks for sharing this. Great job!

hoschiking, 2024-08-01:
@hendrikhoffmann 
what´s the right way to compile your fork of the project?
the original project growatt2mqtt I compiled and it works fine.
I´m not able to do this. 
I have the impression that some code is missing.
Do I have to replace parts of the original code with your code?
It woukd be finem if you would desribe it step by step.
Thank you in advance

hendrikhoffmann, 2024-08-03:
Thanks for letting me know & Fixed.
Either update the project or comment out 
//#define useModulPower 
in settings.h

hoschiking, 2024-08-08:
Your bugfix is working, now I´m able to compile it. thanks !!!
your fork would be absolutely perfect if the MQTT configuration was not implemented in the code, but could be configured via the Web API. Like with Wifi.
If the environment changes, you have to recompile every time. 
It would be fine :-)