nygma2004/growatt2mqtt #2: Getting Error: Invalid slave ID

SHODK, 2022-02-05:
Hi,
What a great project you have made here, I follow you video and think I made a correct setup, however I get Error: Invalid slave ID on the MQTT
My Growatt is same as yours, MIN 3000TL-XE
I check the COM ID on Growatt it is set to 001
I try change it to 002 or 003 with the same change in Arduino sketch
I double check my wiring of Wemos D1 and TTl to RS-485 module, I have same 230V power supply as you show
No matter what I try it give me same error, do you have any idea where to look?

[image]

[image]

nygma2004, 2022-02-05:
OK, so this is interesting. In the log I can see messages saying that "Data MQTT Sent". So it is able to read data off the inverter from time to time. It is possible that the read rate is too quick for the inverter. That does not explain why it would send and invalid slave ID, but still. Can you change the UPDATE_MODBUS variable from 2 to for example 5 and test again?

SHODK, 2022-02-05:
Thanks for the quick reply, it did not help increasing it to 5 or 10, however I did find the fault and it was my own, when compiling the Arduino sketch a few days ago, it gave an error on "mqtt.setBufferSize(1024)" so I commented it out, it turnout that my PubSubClient library was a old version 2.7.0, upgrading to 2.8.0 it could now compile without error, uploaded program and everything works now.
Again, thanks for a great program and video

nygma2004, 2022-02-06:
Oh, wow, I did not know that the setBufferSize was only implemented in 2.8.0. Good catch. I needed this as the data payloads are very big.

closed 2022-02-06 by nygma2004

unimatrixZxero, 2022-08-21:
Adding a data-point here. I had that error as well initially, I noticed that one of the connections to the RS485-TTL wasn't very good. Fixed the connection and the issue resolved itself.  Specifically it was the DE -> D5 connection ( RS485-TTL -> ESP32 D1 Mini)

no-Legend, 2023-07-27:
I'm facing the same problem with my MIC600, all the time I got that error.
Allready using the 2.8.0 version of PubSubClient and there is now warning during the compilation.

nygma2004, 2023-07-27:
@no-Legend sorry but to be exact, are you getting the error on compiling the sketch?
Can you paste in the error that you get?