nygma2004/growatt2mqtt #3: Max485 <----> ESP / IOs are not 5V tollerant

Stumpf74, 2022-03-09:
Hello,

I have a question about the hardware.
you use 5V for your circuit. The ESP generates the 3.3V for the core voltage from this. This means that you would actually need a level shifter between ESP and MAX.

Will it work without one?

Thanks

BR
Stumpf

nygma2004, 2022-03-09:
Yes it certainly works. To be honest I never realized that MAX is 5V. I copied the hardware from another project and used it in many different project, and never had any issues.

Stumpf74, 2022-03-10:
Okay then, I will try this!

Thanks for the project. My Growatt MIN 3600TL-XE I will report briefly on how it works.

Stumpf74, 2022-03-15:
Hello, so it works.

I use the USB connection as the power supply.
This gives me the data I need for my control system.

Thanks for your advance work.

Greetings

closed 2022-03-15 by Stumpf74