nygma2004/growatt2mqtt #4: Wifi does not connect

Alexbuehs, 2022-06-14:
I am a newcomer to the field, I flashed the ESP with the code via Arduino. I made the setting under Settings.h but unfortunately the ESP does not connect to the Wifi after flashing. Can you maybe help me and tell me where exactly I have to set what in the Settings.h tab?
[image]

nygma2004, 2022-06-14:
You need to change all the x-es to the actual values in your system: SSID, password, MQTT server IP address, userid and password.

Alexbuehs, 2022-06-14:
[image]

nygma2004, 2022-06-14:
You can leave it as it is. The ESP will just get a dynamic IP address, it will not impact how it works.

Alexbuehs, 2022-06-14:
First of all, thank you very much for your help. That's exactly how I did it and the esp was flashed without any problems but when I then open the serial monitor there is always WIFI Connecting .............. and that repeats itself again and again. If the esp is flashed, the blue LED also lights up permanently, is that correct?

nygma2004, 2022-06-14:
Yes, the blue is on until it connects to the wifi, and from that point it only flashes when reads the inverter. Can you screenshot the current settings.h file (obviously blurring out the passwords) and the full serial log? Maybe that will give me some ideas on what is going on.

Alexbuehs, 2022-06-14:
I will take a screenshot of settings.h tomorrow and post here. Thanks again for your help.

Alexbuehs, 2022-06-15:
Hey, I realised that the code is running. I deleted your code from the computer again. Then I re-downloaded your code again in Arduino opened the x-es replaced and flashed. The ESP now connects to the Wifi. Now I just have to see that I can do it with the broker. Do that with a raspberry and the iobroker but in the data point it only says true. Is there another way to even see if the esp receives data? I connected all this to a Growatt TL-X.

Alexbuehs, 2022-06-15:
And one more question. Does the Wifi Shine X Stick have to be unplugged?

nygma2004, 2022-06-15:
Check the serial debugger. If it is not able to read the inverter, you get some error messages (like No Response or similar). So if you see no error, the modbus is working.
I think the Shine X plugs into a different port, so you can leave it. I don't have that.

Alexbuehs, 2022-06-15:
How do I get the serial debugger via wifi? Or leave the esp connected to the pc and then via the serial monitor from Arduino?

nygma2004, 2022-06-16:
Yes, use it connected to the computer and user the serial monitor from Arduino IDE.

closed 2022-06-16 by Alexbuehs