nygma2004/growatt2mqtt #5: MQTT data not received from inverter.

Flintebassen, 2022-07-07:
Hi.

Since this morning i only receive data on Growatt/status not on Growatt/data.
I have tried both the 1.2 and 1.3 code.

Maybe it's just my inverter, i dont know.
It have been running fine for 1 month, but stopped this morning.

MQTT explorer also only shows data from Growatt/status.

[image]

[image]

nygma2004, 2022-07-08:
The "Error: Response timed out" is an error message from the modbus communication. For some reason your ESP is no longer talking to the inverter over modbus. I am not sure what the issue can be. I recommend to check the wiring and the connection first. Maybe one of the wires somewhere got loose.
My second guess that the modbus module has died, but it never happened with me so that I think is less likely.
Nevertheless, please use the same firmware that was working before.

Flintebassen, 2022-07-08:
Hi, thank you for getting back to me.
I did not consider wiring because i get the data from Growatt/status fine.
I will check it anyway.

Flintebassen, 2022-07-08:
I have checked the wiring, all ok.
I will try and power down/up the inverter tomorrow.
The problem exist on any of the firmware. I only tried the newest one because the old one did not work anymore.

nygma2004, 2022-07-10:
I am not sure what to say. The client address of the inverter is 1, and I don't think it can even be changed (or how). So I really don't know why it is not communicating. Again, all I can think is that the TTL to RS485 module died.

Flintebassen, 2022-07-14:
You where right. The RS485 adapter was faulty. I replaced it with another brand, and it is working now. Sorry for the confusing.

closed 2022-07-14 by Flintebassen