nygma2004/growatt2mqtt #7: Publish Home Assistant auto-discovery sensors

That-Dude, 2022-08-14:
First of all I just want to say thank you for this project. I've got it working with my Growatt inverter and learned so much in the process.

I use Home Assistant and the trickiest part was figuring out how to import the values, I achieved this by adding MQTT sensors to my `configuration.yaml` but it's a hacky solution.

```
sensor:
- platform: mqtt
  name: 'Growatt Solarpower'
  state_topic: 'growatt/data'
  value_template: '{{ value_json.solarpower }}'
  unit_of_measurement: 'W'
  device_class: power

- platform: mqtt
  name: 'Growatt Enery Today'
  state_topic: 'growatt/data'
  value_template: '{{ value_json.energytoday }}'
  unit_of_measurement: 'kWh'
  device_class: energy
```

I was wondering if you might add the functionality to publish the sensors from your sketch into the `homeassistant/sensor/growatt' topic, as this would allow Home Assistant auto-discovery:

I borrowed this example from another solution: 

```
Topic :homeassistant/sensor/inverter_1_pv_power/config

{"device":{"ids":["sa_inverter"],"mdl":"Growatt MIN/MIC","mf":"Growatt","name":"Growatt MIN/MIC"},"device_class":"power","name":"PV power","state_topic":"solar/inverter_1/pv_power/state","uniq_id":"inverter_1_pv_power","unit_of_measurement":"W"}
```

nygma2004, 2022-08-14:
Hmmm, I have seen these homeassisant topic in my MQTT server, but never really paid much attention to them. But it does not look too complicated. But it will be a hell of a large big configuration message considering the high number of fields. I am in a middle of a large project, but I have some tidy up work to do here, so I think I will come back to this somewhere during the fall and add the homeassistant part as well.

That-Dude, 2022-08-16:
ok that sounds great, I'm looking forward to seeing the results.

I'm super pleased with your sketch, it's been working flawlessly for a week now.

nikito7, 2022-09-13:
For large configurations yaml it's easier to manage

jtauscher, 2023-01-26:
> For large configurations yaml it's easier to manage.
> 
> One advantage of discovery, it's create a device on HA

> For large configurations yaml it's easier to manage.
> 
> One advantage of discovery, it's create a device on HA

Hi nikito7, can you please explain how it works? Do you maybe have an example yaml code?

nikito7, 2023-01-26:
https://github.com/nikito7/easyhan/blob/dev/growatt/pv1-discovery.txt