nygma2004/growatt2mqtt #11: To obtain the data of exported energy.

josemalm32, 2022-10-16:
Hello, I wanted to make a request to see if it is possible. I would like to retrieve the data of energy exported to the grid or otherwise the energy imported from the grid. In my case I have a meter connected to the growatt inverter, since it is obligatory in Spain.

[image]
[image]
[image]

I can see that data on the web, but I can't figure out what it is in modbus. How would it be possible to obtain this information? Thank you very much.

nygma2004, 2022-10-16:
Well, I found the modbus documentation for my inverter on the internet. From the above screenshots I can't tell what type of inverter you have, and I have no idea whether the documentation is available for that online.

josemalm32, 2022-10-17:
Hello, thank you very much for replying. My inverter model is Growatt MIN 4200TL-XH. Theoretically it already uses the RTU Version II protocol version. You can see it here. https://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf .But I am not very clear which can be the code that corresponds to the energy that is exported.

nygma2004, 2022-10-18:
Well, it is not clear for me either. But I think you need to try with some of the high registers and see they provide and useful information. Like Input register 1015-1016,  1021-1022, 1023-1024, 1029-1030. This is my best guess.

dst10, 2023-03-13:
Hello, did you finally figure out the correct modbus address? I have the same problem

TBierstedt, 2023-04-12:
It seems like you're using the Eastron SDM230 for single-phase energy measurement. I found the following document for it: https://bg-etech.de/download/manual/SDM230-register.pdf. The registers start at 30001 in this document, which might be helpful. Unfortunately, I don't have any knowledge of Modbus or programming, so I can't be of further assistance.