nygma2004/growatt2mqtt #16: use  growatt2mqtt  on classic arduino

geimel, 2023-01-28:
Hi,
I've found your great project and will use a part of your code on a arduino nano, to display the data on a oled display.
Yesterday I've received my MIC 1500TL-X, I have connected the RS485A1 and B1 to an usb-rs485 convertert, I've send several requests (read input an holding register) but get absolutely no answer.
Is it possible to help me with an example complete modbius request message, so I can send this over a terminal an check the connection.
(the RS485A2 B2 port works well, here I can see the requests of the inverter to smartmeter after activating this, so I think the USB Converter works)
Thanks a lot 
Richard

nygma2004, 2023-01-28:
OK, so you have connected the inverter with your laptop with a usb-rs485? What program do you use to read data? What do you mean that you get nothing?

geimel, 2023-01-29:
there is no response from the MIC when I send a query

nygma2004, 2023-01-29:
Well, it is hard to tell from here. Few obvious questions:
- MIC is running/powered?
- Did you set the communication to 9600 baud, 8, N, 1? Slave ID = 1? Did you pick the correct COM port?
- Try swapping the A and B wires
- The USB and serial is close to the inverter?
- You are reading input registers, right?
- You are reading a few, like address 0 quantity 5?

geimel, 2023-01-30:
> Well, it is hard to tell from here. Few obvious questions:
>
>     * MIC is running/powered?
>
>     * Did you set the communication to 9600 baud, 8, N, 1? Slave ID = 1? Did you pick the correct COM [...]

All above = yes.
I think it is a hardware defect :-(

nygma2004, 2023-01-31:
That is also possible. It would be a good idea to test with another USB to RS485 adapter.