nygma2004/growatt2mqtt #20: MQTT "error" topic is not cleared after succesful transmission

matthias-bs, 2023-03-11:
Hi Csongor,

if there was a Modbus communication error, e.g. `error = Response timed out` and communication is resumed normally, the MQTT topic "error" is not updated.

I think in `ReadHoldingRegisters()`, the "error" topic should also be updated if the condition `(result == growattInterface.Success)`  is true (e.g. error = "none").

Besides, the message string should be put in double quotes, i.e. `error = "Response timed out"`.

Regards
Matthias

nygma2004, 2023-03-13:
Hi Matthias,

I was not planning to update the error topic with anything, after all once communication is up and running, you will start receiving messages in the data topic. Therefore the error topic keeps the data of the last reason the communication failed.

I am not sure what do you mean with double quotes, the message text is already in double quotes in the code.

Csongor

matthias-bs, 2023-03-14:
Hi Csongor,

You could have sporadic Modbus transmission errors, even after the communication was set up correctly.

The current output is `error = Response timed out`, but it should be `error = "Response timed out"`, because the value is a string. (please compare to the other MQTT topics with strings) Most MQTT clients/JSON parsers seem to accept the former, but it could lead to problems with others.

To achieve this in the code, you can write
```
message = "\"" + message + "\""; 
mqtt.publish(topic, message.c_str());
```

Best regards,
Matthias