nygma2004/growatt2mqtt #22: RS232 Viability

chromos33, 2023-03-18:
Let me preface this with saying I am an absolute beginner on arduino etc. (webdeveloper with some c# experience)

But would it be possible to adapt this library to use this over the usb connection on the MIC 600 TL-X or do I need the RS485 cable connection?

As far as I can google the usb is apparently just an RS232 wrapper.

chromos33, 2023-03-18:
nvm figured it out by referencing https://github.com/Lyve1981/GrowattUsbModbus
just need to replace the conents of growattInterface.cpp growattIF::initGrowatt()
with 
Serial.begin(115200);
  growattInterface.begin(SLAVE_ID , Serial);

  static growattIF* obj = this;

closed 2023-03-18 by chromos33

nygma2004, 2023-03-19:
Sorry the late response. I am glad you managed to figure it out, because honestly I would have no idea how to do this. Kudos.