nygma2004/growatt2mqtt #23: useModulPower

chromos33, 2023-04-01:
I just tried to set the ModulPower only get a response of Illegal Function.
Question is what Firmware version does this work with. 
Mine is v1.2.0p2s

just to make sure it's the fact that I use this over the usb interface instead and not just growatt locking the function in a more up to date firmware.

nygma2004, 2023-04-02:
@joe288 could you please check this? I did not use the ModulPower function yet.

joe288, 2023-04-03:
Yes, from version v1.2.0p2s this is integrated.
To be on the safe side, the function must first be activated in settings.h by removing the comment 
before **#define useModulPower 1**. 
Otherwise the function is not known.

chromos33, 2023-04-03:
Hmm ok possibly the setting did not take or it just sometimes responds with illegal function.

Works now thanks.