nygma2004/growatt2mqtt #25: Use growatt2mqtt for evcc

aherzer69, 2023-04-07:
Hello,

I would like to use the data from growatt2mqtt for EVCC 

EVCC propsed following for Growatt inverter:
([https://docs.evcc.io/docs/devices/meters](url))

`meters:
  - name: my_grid
    type: template
    template: growatt-hybrid
    usage: grid
    
    # RS485 via adapter (Modbus RTU)
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0 # USB-RS485 Adapter Adresse
    baudrate: 9600 # Prüfe die Geräteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter für den Adapter
    
    # RS485 via TCP/IP (Modbus RTU)
    modbus: rs485tcpip
    id: 1
    host: 192.0.2.2 # Hostname
    port: 502 # Port
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.0.2.2 # Hostname
    port: 502 # Port`

Is it possible to implement here the growatt2mqtt data?

or

I found also an older generic format of evcc:

`meters:
  - name: my_solarlog
    type: custom
    power:
      source: calc
      add:
        - source: modbus
          uri: 192.168.0.77:502
          id: 1
          register:
            address: 3502
            type: input
            decode: uint32s
          scale: -1`

As I understood, this is refering directly to modbus data.?
But I don't know how to use. 
Is the "uri" the IP of my Inverter or from D1-mini with growatt2mqtt?
"id" = ?
"address" = modbus id of Growatt protocol? ("1" for "Input power (high)", ...)

I am sure it is possible, but I was not able to solve.

nygma2004, 2023-04-07:
OK, after reading about this page for 15-20 minutes, it looks like this software can read modbus directly. Ideally you don't even need my converter at all. You can plug an RS485 to USB converter into the computer/SBC this software is running on and connect the A/B lines to Growatt directly. And it will read the data straight from the inverter. You can see Growatt examples in their page. Remember RS485 can travel few hundred meters without any problems. 

Or you can get a modbus RS485 to TCP gateway, and use the TCP example. I would try either of these routes and it will be a direct communication between evcc and the inverter.

As far as I can see, these examples support some generic Growatt Hybrid inverter and TL-XH series, so if you have one of those it will work. But they don't seem to have a template for other models.

aherzer69, 2023-04-08:
Thank you very much for effort and quick reply!!!
I am very happy with your software! It is running since weeks w/o problems.
With your software I have robust values (just in time!) to use it for my Home Assistant activities.
If it is not possible to use the access of growatt2mqtt to evcc, I will try to use the mqtt data in evcc.

nygma2004, 2023-04-09:
I don't think it is possible. I have seen modbus client implementation for the ESP, but a modbus master.
The only option I can think about is a modbus RS485 to TCP gateway and through that evcc and homeassistant could read your inverter as well.