nygma2004/growatt2mqtt #27: Growatt MOD 6000TL3-X

Teppis88, 2023-04-16:
Have to say that this code works fine with this 6kw 3phase model. Only problem is that I can't get voltages from different phases.

Can I somehow scan all the messages that growat sends so I could check em if it sends them?

nygma2004, 2023-04-16:
Yes you can. To make your life easier, maybe you can get a RS485 to USB dongle and you can access modbus directly on your laptop. It is easier to change the modbus parameters in a modbus tester program like Modbus Poll than changing the code, compile and upload. Just like other voltages, the voltage is stored on 0.1V, so you should look for values in 2200-2500 range.

joe288, 2023-04-22:
You can query additional values ​​in the **growattInterface.cpp** in the **ReadInputRegisters()** function.
As an example you can see how **gridvoltage** is implemented.

[image]

I think the values ​​you should be interested in are in the area.
[image]

You have to test the value of the information that interests you.

Unfortunately, I only have the single-phase version and can't do it for you.

joe288, 2023-04-29:
see changes from here https://github.com/wroadd/growatt2mqtt_for3phases

Teppis88, 2023-05-01:
Thanks Joe288! That works :)

closed 2023-05-01 by Teppis88