nygma2004/growatt2mqtt #28: Sending setMaxOutput to MQTT

sstolk90, 2023-04-21:
Hi,

I would like to try this setting to reduce the power of the inverter in case of a negative energyprice when my battery is full.
In MQTT Explorer I tried to send this:

[image]

[image]

What am I doing wrong?
What is the syntax for the MQTT message?
I can not find any documentation in the code.

nygma2004, 2023-04-21:
Let me add @joe288 as he added this functionality to my original code.

sstolk90, 2023-04-21:
Since when is this in the code?
Because I think it's not in mine...

joe288, 2023-04-22:
Hello,
the value is simply transferred in decimal. 
From 1-100%. 
[image]
You can also switch off the Growatt completely via setEnable.
[image]

sstolk90, 2023-04-22:
Like this: 
[image]
I get not any confirmation and the Growatt is still producing more then the limit. 
Is this implemented after 20 aug 2022?

joe288, 2023-04-22:
this was added to the master branch of nygma2004 on 10 December.
[image]

sstolk90, 2023-04-23:
Thanks! So it is not in my sketch yet and that's why it won't scale down when instructed to.
I cannot update the sketch since Arduino IDE is bugging me big time but that is not part of this issue.
I get lots of errors for a FastLED and a lot more of definitions from certain files he cannot find.

Issue can be closed and thanks for your help!

closed 2023-04-23 by sstolk90