nygma2004/growatt2mqtt #29: my version

Performance20, 2023-05-11:
Hi,

I forked your software and made some changes. I saw your comments, if stabillity is no so good and OTA firmware update not working
- I migrated to platformio/VSC
- change the string handling which add control of the size of the used buffer (e.g. sprinf to snprintf), which avoid buffer overruns. And there was realy 1 value which was one letter to long for the used buffer. Also using the same buffer as input and output in sprinf funktion could work but the statement is here "behavior not defined". Therfore I changed it.
- Also I removed all this MQTT building code out of the timer funktion. I thing, interrupt routines should be short and fast as possible and not lead to cascading of interrupts
- I added one feature. Now it is possible to change the update times for sending the MQTT messages (original fix values in the code via #defines). Now changeable via MQTT messages and stored in the EEprom of the ESP.
- and some other smaller changes
- But I removed all this RGB led stuff     

I have tested it with my Growatt, run fine, Firmware update via OTA works fine well.

Helge

nygma2004, 2023-05-14:
Hi Helge,

Thanks a lot for your work. I know I should really switch from the Arduino IDE to VSC. Everyone says that it is so much better. But I think what would be even more helpful for the community if I would add WifiManager, so I can offer a binary that can be easily flashed to an ESP and WifiManager can be used to configure the server and modbus settings.

I have to be honest, I don't quite understand the importance between sprinf and snprintf, but I will investigate and learn about this more.
Moving most of the code out of the timer callback is a good idea. I was concerned in the beginning that it will cause issues with watchdog timer or something, but since it ended up working, I did not bother to review it.
Setting the refresh interval via MQTT is also a nice feature, and the RGB if only secondary for the entire function of the code, so not important.

I will share you project on my main page.

Regards,
Csongor

Performance20, 2023-05-26:
Short update from my site:
I removed the hard coded Wifi credentials from the code and inplemented a captive portal by using ESPConnect lib.
If the ESP is not configured to Wifi, he starts in AP mode and provide such a web portal, where the SSID can be selected and passeword enterd. Credential will be stored in eeprom.

Regards

Helge

Diddlik, 2023-05-26:
> Short update from my site: I removed the hard coded Wifi credentials from the code and inplemented a captive portal by using ESPConnect lib. If the ESP is not configured to Wifi, he starts in AP [...]

Is it possible to move mqtt setup in web settings too?