nygma2004/growatt2mqtt #31: Smart Meter emulation ingtegration

Diddlik, 2023-05-26:
Hi guys, since begin the use of Growatt I have built myself a small USB dongle to read growatt (https://github.com/Diddlik/USB_Dongle_ESP8266_Max485). 
Now I plan to install export limit as well. A good project is here: https://github.com/alessandromatera/SDM230-emulator-growatt-esp-mqtt

What I want to achieve is that I have a stick, an ESP, which is connected to two RS485 converters.
One is used to read Growatt (pin 3/4) and the other Converter would be connected to pin 7/8 to act as a smart meter. 
Is it possible to combine the Code of alessandromatera and yours?

The development of hardware part seems to be an easy step, but software part imho not so easy...

nygma2004, 2023-05-30:
As long as Growatt and the smart meter use the same communication speed and settings (9600, 8, 1, N) and the two devices are set to different slave ID, it should work. I would reassign either Growatt or the smart meter from the default slave ID of 1 (I presume) to a different address. And you just need to read / write the smart meter and Growatt after each other. The current Growatt logic already has the SetMaxOutput implemented. 

But looking at the alessandromatera code, it does not talk to a Modbus smart meter, but it uses a Shelly smart meter and reads the data over MQTT. So it is not exactly doing what you plan to implement.

Diddlik, 2023-05-30:
> As long as Growatt and the smart meter use the same communication speed and settings (9600, 8, 1, N) and the two devices are set to different slave ID, it should work. I would reassign either [...]

I am a little afraid that SetMaxOutput write to EPPROM from the inverter, which is not good in due long term.
The connection with second RS485 as smart meter solves the problem, because inverter dynamically controls it by itself without writing it to flash every time. or have I thought wrong here?

nygma2004, 2023-05-31:
Possible, but I have no way of telling how the inverter works internally. But both sketches write to the modbus register, therefore I don't see how the SetMaxOutput in our code and the same modbus code in alessandromatera example would be any different.