nygma2004/growatt2mqtt #32: Growatt MIN 3600TL-XE ?

newegnaro, 2023-06-05:
Hello, 
Thank you for sharing this project. I recently has a solar system installed with the above Growatt inverter and I'm trying to get basic data from it into Home Assistant (HA). I've tried some of the HA integrations that pull from the Growatt server but these suffer from being blocked. So I thought I might be able to pull the data from the Modbus registers.

Your project looks like it could work if I pull the data from MQTT, however the SYS-COM definitions in the manual for my inverter  differs from the one you show. (see following image of manual)
[image]
Do you think it might work ? which of the ports from sys-com should I use ?
Sorry for the simple questions, very much of this is new to me.

nygma2004, 2023-06-07:
I think the smart meter connection is used to connect a meter so the inverter can limit the production to match your consumption (not to sell electricity to the grid). So it is definitely not that one.
I think it should be a the Connect Min ShineBus connection.

newegnaro, 2023-06-07:
Thank you for your thoughts. I need to decide if I have the skill to solder up your project to try this way. OI have been prototyping via a breadboard and ESPhome... but no luck yet.

newegnaro, 2023-06-20:
Hello, thank you for pointing this out. This was the thread i was using for my breadboard example but no luck despite trying a few of the code samples. I might have to try the flashed standard dongle method. Which inverter do you have please ?

nygma2004, 2023-06-20:
I have a 6 year old 3000-TL, and I also have a MIN 3000 TL-XE and MIC 1500 TL-X. The MIN and MIC models have the same connector (as documented in this project), the older 3000-TL has an old style 9 pin D-SUB9 connector.