nygma2004/growatt2mqtt #34: Used your code! Thanks! Reworked main as code seems to freeze

matthijsfh, 2023-11-13:
Sorry, no way to contact you directly except via a issue ticket ;-)
Thanks for reverse engineering the inverter. Works like a charm.

However the "main" code seems to freeze.

* I think  **while (!mqtt.connected())** is the root cause if the wifi is lost after the last wifi check. This code will never exit the loop, as the mqtt will not connect, because the wifi is lost.

* Added all relevant parameters to the wifi manager (when then saves them to flash). Having to re-compile to change mqtt server settings is not my preferred solution. I have had issues with this on another ESP project. After 3+ years a mod was needed, and it did not compile anymore due to "progress" in the development tools.

* Added jumper to force the wifimanager to show.

* Added a FAN to cool the inverter which uses the temperature from the electronics board inside the inverter to switch on/off.

Can send you my changes if you are interested.

Greetings Matthijs

nygma2004, 2023-11-14:
Hi Matthijs,

Oh, these are great changes. I am still using my much older version of the code before it got re-worked by Joe. Wifimanager is great, actually I should add it to more of my projects. This also means that we can also offer complied binaries to users which is much easier to flash than if you have to build it yourself.
Good catch on the mqtt.connected issue. I never noticed this problem, I shall look at my version if I had done something differently. A few months ago my laptop died and I have not installed the IDE on my new device (was not working on project recently).
The jumper for the wifimanager is also a good idea. These sequence issues are a real pain to find, mostly because you cannot think of all these cases until it happens. Nowadays I am batteling with some modbus issues between my Node-Red and PLC which all of sudden started happening a few days ago. No software changes, nothing.
Fan is also a great idea.
I am interested in your implementation. If you send it, can you also send me a photo how the jumpe and the fan connection is implemented. I will add them to the readme for others to use it as an example.

matthijsfh, 2023-11-14:
Hi,

I will share my code with you. Think I found your email somewhere ;-)
My growatt code is stored my private github with much more projects.

Did remove some of your features, because I did not want to damage my inverter by sending some commands to it.

Anyway, have a look and maybe you can use some features of it.

Greetings Matthijs