nygma2004/growatt2mqtt #35: "PV energy today" and PV energy total missmatch

Kletternaut, 2023-11-18:
Hi Csongor,
at first i want to thank you very much, for your work here and also on Youtube. I'm on of your subscriber and have to say, i have learned a ton from what you doing and explaining... in an age of >55 ;-) Thank you man!

So now, i want to inform you there is a register mismatch and failure in the code For MIC-TL-X.

I do not get total- or today-energy values in mosquitto or influx. I've checked the register with the [Growatt Inverter Modbus RTU Protocol V.1.2](https://github.com/nygma2004/growatt2mqtt/blob/main/Growatt%20PV%20Inverter%20Modbus%20RS485%20RTU%20Protocol%20v120.pdf) and it says on the very first page:

> Instruction: Register range for various types of inverter
>
>**TL-X** (MIN Type): 03 register range:       0-124, 3000-3124; **04 register range: 3000-3124**, 3125-3249
>...
>Storage (MIX Type): 03 register range:  0-124, 1000-1124; 04 register range: 0-124, 1000-1124
>...

So, there is **no input** register type range (**04**)  **0 - 124** in the **MIC TL-X** devices, only **3000-3124** also **not even** the (04) 3125-3249 range like the manual says.

your code: growattinterface.cpp:

> modbusdata.energytoday = ((growattInterface.getResponseBuffer(**53**) << 16) | growattInterface.getResponseBuffer(54)) * 0.1;
> modbusdata.energytotal = ((growattInterface.getResponseBuffer(**55**) << 16) | growattInterface.getResponseBuffer(56)) * 0.1;
> modbusdata.totalworktime = ((growattInterface.getResponseBuffer(**57**) << 16) | growattInterface.getResponseBuffer(58)) * 0.5;
> 
> modbusdata.pv1energytoday = ((growattInterface.getResponseBuffer(**59**) << 16) | growattInterface.getResponseBuffer(60)) * 0.1;
> modbusdata.pv1energytotal = ((growattInterface.getResponseBuffer(**61**) << 16) | growattInterface.getResponseBuffer(62)) * 0.1;

The Growatt RTU Protocol Description for MIC-TL-X input (04) registers says:

**3053** | Epv_total H | PV energy **total**
3054 | Epv_total L  | PV energy total
**3055** | Epv1_today H  | PV1 energy **today**
3056 | Epv1_today L  | PV1 energy today
**3057** | Epv1_total H  | PV1 energy **total**
3058 | Epv1_total L   | PV1 energy total
 
Could you maybe check and correct this?

Greetings from Karlsruhe/Germany
Tom

nygma2004, 2023-11-18:
Hi Tom,

I also have a MIC-TL-X and the code works just fine there. There is a naming convention in modbus that the input registers are always number from 3000 and the holding registers from 4000 (or the other way around, I can never remember). This is done so when you look at the number you can immediately tell if it is a holding or input register. But in all the modbus software I used and also in the arduino library, you provide the address from the base 3000 or 4000. So what is addres 3000 in the documentation, in the program that is zero. 
As I said this code is reading my small MIC 750 and it is fine.

Kletternaut, 2023-11-18:
What you write makes perfect sense, but various totay and total values are still mixed up. Take a look above.
What irritates me now is that I can read out the total and today data via the 3049 with my own nodered flow (attachment) over an waveshare modbus-wlan converter. But not with this esp-script and your Grafana dashboard. All other data are coming in.
[image]

Kletternaut, 2023-11-18:
I have checked the mqtt-mousquitto log file. `{"fields":{"deratingmode":0,"energytoday":0,"energytotal":0.9,"faultbitcode":0,"faultcode":0,"gridfrequency":49.99,"gridvoltage":227.2,"ipf":20000,"opfullpowe
r":0,"outputpower":21.5,"pv1current":0.3,"pv1energytoday":0,"pv1energytotal":0.9,"pv1power":22.1,"pv1voltage":58,"pv2current":0,"pv2energytoday":0,"pv2energyt
otal":0,"pv2power":0,"pv2voltage":0,"realoppercent":0,"solarpower":22.1,"status":1,"tempboost":0,"tempinverter":26.5,"tempipm":26.5,"totalworktime":165833.5,"
warningbitcode":0},"name":"mqtt_consumer","tags":{"host":"raspberrypi","topic":"growatt600/data"},"timestamp":1700298901}
`
Any ideas?

Kletternaut, 2023-11-19:
Hi Csongor,
i have checked it now multiple times. I am very sorry, also because i know that you have much more knowledge of the matter than i do. I have to say your code is not using the correct registers for the TL-X Series, because you've used the wrong register list documentation.
>There is a naming convention in modbus that the input registers are always number from 3000 and the holding registers from 4000

I noticed this for some manufactures (i believe the Eastron SDM630/230) with 3000x (for holding) and 4000x (for input), but now after my review i know it has nothing to do with this. These in the docmantation described 3000-registers are the right, real and only ones that worked correctly. You can't use register 0001 from a other device alternatively to the 3001 and so on. It is different address with different values in it, even when you get some values out of it and even many of them seems to be the right ones. 

The documenatation (site no: 48) says it very clear: The input register for TL-X begins at 3000-3124, (3125...  right after the heading information:
>**Use for TL-X and TL-XH**
and this means **do not start at site no: 33 with input register no: 0.** of an different device.  In many cases the register are kind identical, but a few are different and others do not even exists. If you use 0001 it will bemapped to 3001 or vice versa.

I say it again: there are no input-register 0-124 for MIC-TL-X described in the documentation.

One example (then i'm done here):
In growattInterface.cpp @line 82:

>modbusdata.pv1energytoday = ((growattInterface.getResponseBuffer(59) << 16) | growattInterface.getResponseBuffer(60)) * 0.1;

Here you want to get value of "energy PV1 of actual day", like it is described in input-register list no. 59, right? But instead it reads (with, or without the internal mapping function) the register 3059 | E**pv2**_today. There is no PV**2** in a MIC-TL-X. And that is the reason why i am getting no value.  I've found a few other mismatched addresses in your code. No question, it is a total mess what they have done in this documantation.

nygma2004, 2023-11-21:
OK, I see the problem. The issue is that it will be very difficult to make code changes without having the actual device and testing how it works. Also I am not really sure what is the best way of making the code work for the "old" devices and the MIC-TL-X as well. Maybe this new mapping is something they will roll out for all new models in the future?

Kletternaut, 2023-11-21:
i have you send an email