nygma2004/growatt2mqtt #36: Monitor multiple inverters? Multi-phase, parallel installation.

RoganDawes, 2023-12-29:
I am trying to help a friend monitor his inverters (5 x SPF5000TL HVM-P, arranged as 2 each in parallel on phase A and B, and 1 on phase C) and batteries (4 x Narada 48NPFC100). He has no ShineWifi dongles - apparently these inverters make use of the ShineWifi-F model, which talks RS232 over the USB D+ and D- lines of the USB-A connector.

I have been able to create an ESP32-based device with an SP3232 RS232 transceiver that can talk to one inverter via the USB-A connector. Now I am trying to figure out how best to communicate with the remaining inverters, and the batteries.

With the RS232 connection to a single inverter, and trying to brute force enumerate modbus slave addresses, I was able to find 1 (expected), and 17. I was wondering whether 17 would be the paralleled inverter, 17 == 16+1 or 0b1001, but the readings I got from address 17 were totally inconsistent with the readings from address 1. Lots of 65536 values. Anyone have any idea what might be at address 17?

I'm totally open to switching over to RS485 if it will make life easier in any way. That said, all I have read in the Growatt manuals has not provided any way of changing the modbus slave address from the default 1, which would mean that I would need a RS485 bus for each inverter regardless, to avoid conflict. In which case, since the RS232 transceiver chip I am using has two channels already, and the ESP32 can support up to 3 UARTs simultaneously, I could just use one ESP32 per phase to talk to two inverters, and add an RS485 transceiver to the C-phase ESP32 to talk to the batteries. That said, is it even necessary to query the paralleled inverter, should the readings not theoretically be the same as the primary?

Any other suggestions?

RoganDawes, 2023-12-29:
I did find a mostly unintelligible description of setting the RS485 address on the last page of this document: https://watts247.com/manuals/gw/GrowattModBusProtocol.pdf

> Knock the pv inverter to let the lcd display to the “COM Addr: xxx”, then double knock, if displays “Move”, you should another double knock, until it displays a address number, then you can give a single knock to change the address, this address will be remembered when the lcd backlight off.

Not sure what "knock"ing is in this context?

nygma2004, 2023-12-31:
You literally knock on the display on the unit to change the screen. It has been a long time since I did mine, but you can change a few things through the screen like language, internal clock and the modbus address. Probably there is a vibration sensor behind the screen. A simple finger tap does not always trigger it, but a knock like you knock on the door does.
And yes, with RS485 and modbus you should be able to read multiple devices with a single ESP. Just remember the wire the RS485 lines in series (one device after the other), and not in star shape. And yes you would read each device one-by-one in a loop.

RoganDawes, 2023-12-31:
Oh jeez! That's a super intuitive user interface! Thanks for the info, I'll try see if I can get it to do the necessary!

nygma2004, 2024-01-01:
Yeah I know. I am pretty sure there are robust solution nowadays to waterproof buttons or even use some sort of capacitive sensor. But anyway. Not that I am nocking a lot on this thing...

RoganDawes, 2024-01-14:
Seems like my friend's inverters are newer than the knockable ones. I do appear to have the modbus holding register that controls the inverter's slave address available in the documentation, so my plan is just to change that, and see what happens. If necessary, I will power cycle the inverter, as I have seen suggestions that it ends up responding to commands on the new address, but responding from the original address until after the powercycle!
For what it is worth, I queried the inverter serial number on slave address 17, and got the same serial number as on address 1, so it would appear to be an odd artifact. Thanks for the help. I will connect via RS485 if I am successful at changing the addresses.

closed 2024-01-14 by RoganDawes