nygma2004/growatt2mqtt #37: [Question] useModulPower outside the mqtt context

chromos33, 2024-03-17:
Hey,

I'm trying to use the useModulPower function standalone like plugging in the ESP board like an usb stick and have it rewrite the
Inverter Config.

Tried adding the code to the Setup method after initializing the Interface but it won't update the config in the Inverter. All I'm getting is either illegal function or response timeout. Is there some special sauce that makes this work in the mqtt request or am I just going crazy.

like in the Growatt2mqtt_1p2s.ino (removed mqtt code)

```
void setup()
{
     //Other init code
     growattInterface.initGrowatt();
     growattInterface.writeRegister(growattInterface.regOnOff, 0);
    delay(500);

    result = growattInterface.writeRegister(growattInterface.regModulPower, int(strtol(message.c_str(), NULL, 16)));
    delay(500);
    growattInterface.writeRegister(growattInterface.regOnOff, 1);
    delay(1500);

    if (result == growattInterface.Success) {
      holdingregisters = false;
    } else {
      //using WebSerial to send the result
    }
}
```

Help would be appreciated.

nygma2004, 2024-03-19:
I think the first issue here, that 0A to set the inverter to 1000W is not a string, but a hex value: 0x0A. 
I never used this function myself, let me add @joe288 as he added this part of the code.
@joe288 just to recap, when you publish to the "SetModulePower" topic, you expected to pass the value as decimal and not at string like "0A" right?
Based on this, I think instead of 
```
String message = "OA";
```
it should be:
```
uint16_t modulpower = 0x0A;
```
and further in the code, instead of
```
result = growattInterface.writeRegister(growattInterface.regModulPower, int(strtol(message.c_str(), NULL, 16)));
```
it should be:
```
result = growattInterface.writeRegister(growattInterface.regModulPower, modulepower);
```
Please wait until @joe288 confirms it.

joe288, 2024-03-21:
Hi,
@nygma2004 your change should produce the same result. In the implementation with MQTT it still has to be converted from string to integer. You can optimize this step with the variant of nygma2004.

the code should actually work.
you have to make sure that you first can switch the inverter on/off.
````
growattInterface.writeRegister(growattInterface.regOnOff, 0);    // switch off the inverter
growattInterface.writeRegister(growattInterface.regOnOff, 1);    // switch on the inverter
````
The configuration can only be changed when the inverter is switched off.

Of course it could also be that your inverter has a different timing than mine. I would test to increase the delays significantly to make sure that the commands have been processed completely.

I hope these tips help you to find the problem.

chromos33, 2024-03-22:
thanks for the help, am going to try that next.
Timing thing isn't/shouldn't be an issue as over mqtt it worked even useing R232 instead of what the repo is saying

chromos33, 2024-03-23:
Yeah that was it. thanks again.

closed 2024-03-23 by chromos33