nygma2004/growatt2mqtt #39: Adapting for 3 phase inverter

mintgroen, 2024-06-01:
Hi, 

Great work with all the detailed explanations and even videos!

I have a similar [repo](https://github.com/mintgroen/growatt_mqtt/) which I'm using for quite some years now but since the netmetering policy is going to change I was looking for a way to control the output power. Seems that you have already implemented this :)

I have a MOD 9000 TL3-X so that is a 3 phase model. Since you named your folder 1ps2, do you also have a local 3p2s repo which you can share? Else I am going to fork yours and add the 3 phase registers. 

Thanks,
Maarten

nygma2004, 2024-06-03:
Sorry I don't. I only have single phase inverters, no 3 phase inverter to test with.
If you look at some of the closed issues, this has been asked in the past. Maybe somebody amended the code for 3 phase as well and made it public.

mintgroen, 2024-06-11:
I'm testing on my 1 phase inverter and it looks nice. Great work.

I'll give it a go to upgrade the registers to support the 3 phase models.