nygma2004/growatt2mqtt #40: "maxpower" setting?

matthias-bs, 2024-07-05:
Hi,

let's assume "modulPower" has been changed from "0006" to "000A" and ""maxoutputactivepp" from 100 to 80, resulting in:

``` JSON
{
    "enable": 1,
    "safetyfuncen": 25,
    "maxoutputactivepp": 80,
    "maxoutputreactivepp": 0,
    "maxpower": 600.0,
    "voltnormal": 100.0,
    "startvoltage": 50.0,
    "gridvoltlowlimit": 184.0,
    "gridvolthighlimit": 287.5,
    "gridfreqlowlimit": 47.5,
    "gridfreqhighlimit": 51.5,
    "gridvoltlowconnlimit": 197.8,
    "gridvolthighconnlimit": 250.7,
    "gridfreqlowconnlimit": 47.7,
    "gridfreqhighconnlimit": 50.0,
    "firmware": "GH1.0",
    "controlfirmware": "ZAAA",
    "serial": "",
    "modulPower": "000A"
}
```

Would you have to change "maxpower" from 600 to 800 as well to achieve a maximum grid power output of 800 VA?

Thanks in advance,
Matthias
```

nygma2004, 2024-07-08:
I did not change the maxpower settings on my inverters, so I am not sure. @joe288 can you please advice?

joe288, 2024-07-10:
Hello, 
your settings look good.
I had a similar effect. Apparently it takes a while for the inverter to change the values. 
What is the maximum output that the inverter has achieved in the last few days?

My settings look almost exactly the same.
````
{
  "enable": 1,
  "safetyfuncen": 509,
  "maxoutputactivepp": 100,
  "maxoutputreactivepp": 0,
  "maxpower": 1500,
  "voltnormal": 100,
  "startvoltage": 50,
  "gridvoltlowlimit": 184,
  "gridvolthighlimit": 287.5,
  "gridfreqlowlimit": 47.5,
  "gridfreqhighlimit": 51.5,
  "gridvoltlowconnlimit": 195.5,
  "gridvolthighconnlimit": 253,
  "gridfreqlowconnlimit": 47.7,
  "gridfreqhighconnlimit": 50.1,
  "firmware": "GH1.0",
  "controlfirmware": "ZAAa",
  "serial": "",
  "modulPower": "000F"
}
```

matthias-bs, 2024-07-10:
Hi,
yes, I can confirm that setting "modulPower" works as expected and that "maxpower" changes automatically afterwards.

Thanks @nygma2004 and @joe288 for the great project and for taking care of this issue!

closed 2024-07-10 by matthias-bs