nygma2004/growatt2mqtt #41: Change the publish frequency

hoschiking, 2024-08-01:
I would like to change the publish frequency from 4 to ~30 seconds.
Which code I have to change for this? where is the frequency defined?
thank you in advance

nygma2004, 2024-08-01:
Try in settings.h and change the MODBUS_UPDATE  and change it to 15.

hoschiking, 2024-08-01:
Thank you, it works :-)
one more question
usually I sould be able to change the frequency and another paramters under the topc /growatt/write
the problem is, the folder /write does not exist. do you have any idea why?
I switsched to expert level in ioBroker and I´m not able to creat this folder.
[image]
I´m missing all writable parameters.
[image]

nygma2004, 2024-08-01:
Yes, this project only implements these few parameters to be writable, rest is fixed in the code.